tstana/cubes-fw #21: Solve all TODOs

tstana, 2019-03-14:
Steps to solve this issue:

1. Do a `grep` for the keyword `TODO` in the `cubes-fw/` folder. For example, using the Git Bash command line:
```
grep -rnH "TODO" .
```
2. Address each of the lines returned: either fix or remove.

Zethian, 2019-06-17:
@tstana remaining TODOs are related to your code and ones you have put in, please resolve or instruct if I should try to apply them too.

tstana, 2019-06-18:
Will handle myself (and check your fixes, too).

Thanks for handling them! Which branch are the fixes on?

Zethian, 2019-06-18:
Remove_Ext

tstana, 2022-10-26:
Issue #57 has been opened in relation to the three remaining (at the time of writing of this message) TODOs within `main.c`. It is worth noting that these are potentially not necessary to solve -- but good to be kept in mind.

commit 2356f0ef (2022-10-26): Solve almost all TODOs

Closes #21.
closed 2022-10-26 by tstana